SamZorSec/Open-Home-Automation #1: Feature request

samabsalom, 2016-11-23:
couldn't find an alternative way to contact you.

Love all your examples.  I was wondering if there was a way to add more LEDs to the same file? I have tried altering your files but to no avail

any help you can give would be appreciated
thanks
Sam

closed 2016-11-23 by samabsalom

mertenats, 2016-11-24:
Hi Sam,
It's possible! In this [sketch](https://github.com/mertenats/Open-Home-Automation/tree/master/openhome/sketches/Entrance), two white LEDs are controlled; in this [one](https://github.com/mertenats/Open-Home-Automation/tree/master/openhome/sketches/Livingroom), two RGB LEDs are controlled.

If you need more LEDs, just apply the same logic (2 MQTT topics: state/command, a function to set the new state, a function to publish the new state and handle the command topic in the MQTT callback function).

Sam

samabsalom, 2016-11-28:
Thanks for getting back to me
here is what ive done to upscale if you're interested

https://github.com/samabsalom/nodemcu-MQTT-8ch_relay-home_assistant

Many thanks again! 

Sam

mertenats, 2016-12-02:
Good job!
In your sketch, you can use D1 instead of 5. You also forget to update the header introduction.
```
/*
   MQTT Light for Home-Assistant - NodeMCU (ESP8266)
   https://home-assistant.io/components/light.mqtt/
   Libraries :
    - ESP8266 core for Arduino : https://github.com/esp8266/Arduino
    - PubSubClient : https://github.com/knolleary/pubsubclient
   Sources :
    - File > Examples > ES8266WiFi > WiFiClient
    - File > Examples > PubSubClient > mqtt_auth
    - File > Examples > PubSubClient > mqtt_esp8266
   Schematic :
    - https://github.com/mertenats/open-home-automation/blob/master/ha_mqtt_light/Schematic.png
    - GND - LED - Resistor 220 Ohms - D1/GPIO5
   Configuration (HA) :
    light:
      platform: mqtt
      name: Office light'
      state_topic: 'changeme/light1/status'
      command_topic: 'changeme/light1/switch'
      optimistic: false
```

Sam