SamZorSec/Open-Home-Automation #4: How you do it that nodemcu (esp8266) appears on homekit with HA?

garvarma, 2016-12-13:
Hi,

First of all thanks, about this so detail manual about how to begin from 0 to install and configuring all in raspberry pi.

After reading and watching video, i have a doubt about the nodemcu and how you put in homekit, because in my case i also have few nodemcus on my network, but well i intall on configure homebridge for home assistant, my nodemcu not apprears on homekit.

How did you configure this nodemcu on HA? like sensors? like binary sensor? and only you installed homebridge-home-assistant package? and after all then appeared automatically on homekit?

I think, that also is possible to do it with sonoff or wemos (both are esp8266), really?

thanks

mertenats, 2016-12-13:
Hi,
Thanks for your message. All the lights were working with Homekit; the binary sensors weren't supported at the time. Devices supported with Home Assistant and Homebridge : https://github.com/home-assistant/homebridge-homeassistant

Steps :

- Installation of Home Assistant, [Homebridge](https://github.com/nfarina/homebridge) and the [Homebridge's plugin](https://github.com/home-assistant/homebridge-homeassistant) for Home Assistant
- Initial configuration of Home Assistant
- Addition of the components (MQTT devices, etc.) within Home Assistant (configuration.yaml)
- Initial configuration of Homebridge (config.json, etc.)
- Restart Homebridge
- Create a new Home within Apple Home application, an Homebridge gateway is normally discovered

The full installation is described in my school [report](https://github.com/mertenats/heia/blob/master/OpenHome/Rapport_07_131016.pdf) (in french)

Sam

oriolism, 2016-12-13:
You can integrate almost everything into Homekit if you use MQTT. Check this:
https://github.com/hobbyquaker/homekit2mqtt

For the sonoff you can use this: 
https://github.com/arendst/Sonoff-MQTT-OTA-Arduino

or this with espEasy:
http://xbmcnut.blogspot.co.nz/2016/08/using-espeasy-with-home-assistant.html

Hope this helps.

closed 2017-01-09 by mertenats