SamZorSec/Open-Home-Automation #9: DHT22 MQTT

outrun0506, 2017-01-25:
Hi,

i flashed your Sketch, and it seems to work.

After my setup in Home Assistant, humidity keeps saying unknow.

Heres the Output from the IDE-Serial Monitor:

```
.........
INFO: WiFi connected
INFO: IP address: 
192.168.0.59
INFO: Attempting MQTT connection...INFO: connected
{
  "temperature": "22.80",
  "humidity": "34.10"
}
INFO: Closing the MQTT connection
INFO: Closing the Wifi connection
```

Temperature works fine.

Thanks for help

BR

Lukas

mertenats, 2017-01-25:
Hi Lukas,
Thank you for your message. The output is correct and the MQTT message is received by Home Assistant (it works for the temperature). There is probably a problem when HA tries to extract the humidity from the payload of the MQTT message. I can not test anything now, but you should find some help at the bottom of this [page](https://home-assistant.io/components/mqtt/).

Are you using the same topic for the humidity sensor?

```yaml
sensor 1:
  platform: mqtt
  state_topic: 'office/sensor1'
  name: 'Temperature'
  unit_of_measurement: '°C'
  value_template: '{{ value_json.temperature }}'

sensor 2:
  platform: mqtt
  state_topic: 'office/sensor1'
  name: 'Humidity'
  unit_of_measurement: '%'
  value_template: '{{ value_json.humidity }}'
```

Best regards,
Sam

ssolmaz, 2017-01-26:
Hi Sam,
I'm using this sketch and there is no problem. I have two dht22 sensors.
One of them always shows humidity 1. This sensor is the same in a different sketch.
I guess the humidity sensor is broken.
I would recommend using a different sensor or try it with a test sketch (without mqtt, only serial monitor).

Hi Lukas,
Thank you for this work.
I use Sensor_dht22, light, light_with_brightness sketchs.
I will use other sketch :)
See you...

outrun0506, 2017-01-26:
Hi Sam,

thanks for helping me out on this Topic,

Heres my HA-Config for this Sensor:

```
- platform: mqtt
  state_topic: 'sensor/esszimmer/dht22'
  name: "Temperatur Esszimmer"
  unit_of_measurement: '°C'
  value_template: '{{ value_json.temperature }}'
    
- platform: mqtt
  state_topic: sensor/esszimmer/dht22'
  name: "Luftfeuchte Esszimmer"
  unit_of_measurement: '%'
  value_template: '{{ value_json.humidity }}'
```

In my Mosuitto-DB i have following entry:

`esszimmer_dht22 sensor/esszimmer/dht22 *{"temperature":"18.90","humidity":"40.60"}`

So it looks like your Sketch reports Values from DHT normally but HA is not able to read the Topic.

I'll do some research on this, or do you have a tip to point me in the right direction?

BR

Lukas

outrun0506, 2017-01-26:
Oh my god...

Sorry guys for taking your Time.

I just found the issue -- it was my fault.

I had a missing ' in my sensor.yaml in the state_topic for the humidity. I just looked on it about 1000 times and did not see it...

Really sorry Guys.

@mertenats Thanks for your great work and sorry for inconvenience.

BR

Lukas

closed 2017-01-26 by outrun0506