SamZorSec/Open-Home-Automation #10: esp8266 LED not working

mrpannu100, 2017-02-27:
Hi 
I am new on github and with home assistant. I was trying your led example. I was not able to turn on led with the provided code. Esp8266 is working properly i checked serial monitor wifi and mqtt connection was successful. Just to ensure my esp8266 works properly i have tried some other examples from github with dht22 sensor that was successful. 
I have checked mqtt log file in raspberry pi. I can see successful connections with esp6288.
Adding to that when i turn on led button from home-assistant screen it turns off by itself after 1 sec.
Am i missing something ?
Thanks

mertenats, 2017-02-27:
Hi,
- Which sketch are you using? 
- What's your configuration in HA?
- Can you show me your logs (ESP8266)?

Thanks,
Sam

mrpannu100, 2017-02-27:
Hi Sam
Thanks for you quick reply.
I am using following sketch for ESP6288.
[https://github.com/mertenats/Open-Home-Automation/blob/master/ha_mqtt_light/ha_mqtt_light.ino]

HA configuration.
```
homeassistant:
  # Name of the location where Home Assistant is running
  name: home
  # Location required to calculate the time the sun rises and sets
  latitude: 49.0997
  longitude: -122.6526
  # Impacts weather/sunrise data (altitude above sea level in meters)
  elevation: 9
  # metric for Metric, imperial for Imperial
  unit_system: metric
  # Pick yours from here: http://en.wikipedia.org/wiki/List_of_tz_database_time_zones
  time_zone: America/Vancouver

# Show links to resources in log and frontend
#introduction: 
 

mqtt:
  broker: 192.168.0.3
  port: 1883
  client_id: home-assistant-1
  username: xxx
  password: xxx  
  
light:
  platform: mqtt
  name: Office light'
  state_topic: 'office/light1/status'
  command_topic: 'office/light1/switch'
  optimistic: false

# Enables the frontend
frontend:

http:   
  # Uncomment this to add a password (recommended!)
   api_password: test123
  # Uncomment this if you are using SSL or running in Docker etc
  # base_url: example.duckdns.org:8123

# Checks for available updates   
updater:

# Discover some devices automatically
discovery:

# Allows you to issue voice commands from the frontend in enabled browsers
conversation:

# Enables support for tracking state changes over time.
history:

# View all events in a logbook
logbook:

# Track the sun
sun:

# Weather Prediction
     
# Text to speech
tts:
  platform: google
```
ESP8266 log

INFO: Connecting to mywifi
INFO: wifi connected.
INFO: IP address 192.168.0.2
INFO: Attempting MQTT Connection... INFO: connected
INFO: Attempting MQTT Connection... INFO: connected
INFO: Attempting MQTT Connection... INFO: connected
INFO: Attempting MQTT Connection... INFO: connected

mertenats, 2017-02-27:
Hi,
Your configuration seems to be ok, but your logs shows that something is going wrong because the connection to the MQTT broker is probably lost. Have some problems with your broker? I had this kind of problem with it in the past (Mosquitto, installed with the HA image).

mrpannu100, 2017-02-28:
Hi
My mqtt seems working fine, as i have tested some other examples with dht22 using same configuration for mosquitto. 
I want to show to my esp sketch(mqtt section)
const PROGMEM char* MQTT_CLIENT_ID = "home-assistant-1";
const PROGMEM char* MQTT_SERVER_IP = "192.168.0.3";
const PROGMEM uint16_t MQTT_SERVER_PORT = 1883;
const PROGMEM char* MQTT_USER = "xxx";
const PROGMEM char* MQTT_PASSWORD = "xxx";

is that fine or do i need to modify some other section of sketch?
Thanks

mrpannu100, 2017-02-28:
Hi
I just figured out that i was putting wrong mqtt client id. I chnaged that now everything looks fine.
My problem is solved.
Thanks

closed 2017-02-28 by mertenats