SamZorSec/Open-Home-Automation #12: Flashing Effect

epiller, 2017-06-22:
Hey mertenats,

Is it possible to make this code present the "Flashing" effect in HomeAssistant?
I saw somewhere that off-the-shelf rgb lights that are connected to HomeAssistant can show "Effects" drop-down menu where you can choose effects like Flash, Fader, etc. and I was thinking if it could be possible to do something like that with your code and ESP8266? I'm inexperienced with Arduino/ESP8266 programming, but I'd be willing to help you as much as I possibly can to make this happen. 

P.S. I've been using those shitty IR remote RGB LED controllers from eBay. I then discovered your code and decided to get a few ESP8266s. God damn if this isn't the best thing ever. Thank you so much for this, seriously.

mertenats, 2017-06-23:
@epiller ,
I'm using effects and MQTT Discovery in this firmware: https://github.com/mertenats/AI-Thinker_RGBW_Bulb.

Regards,
Sam

epiller, 2017-06-23:
Hey,

I just took an in-depth look into that firmware. As I'm a novice programmer, I have no idea how I'd go with porting it to this code.
I don't know how complicated is it, but could you add it to this code? That would be absolutely amazing, and I'm sure others would like to have that feature.