SamZorSec/Open-Home-Automation #15: Arduino ethernet

sobiso, 2017-08-09:
I new in arduino world. It's possible to change yours lib/programs from ESP8266 to Arduino with ethernet shield ? I'm very interested update the firmware over-the-air.

mertenats, 2017-08-09:
Hi @sobiso,
I think that the OTA library isn't compatible with the Arduino boards or the internal memory is probably too limited. Why not buying ESP8266 boards? It's probably possible to add an Ethernet port to them and they cost less money.

Sam

sobiso, 2017-08-09:
Good point with adding eth port to ESP8266. I prefer cable connection. It's possible to change Mqtt 'path' without re-programm ESP ?

mertenats, 2017-08-09:
I'm not sure to understand your question. Normally, in all my sketches, the MQTT topics are hardcoded/derived from the device ID inside the source code. Feel free to modify them.

Sam

sobiso, 2017-08-09:
I'm thinking about one universal program ex. switch. So you don't have to set each time new topic and re-program. This topic can be set via http or something like that and save to eeprom. What are you thinking about this ?

Kuba

closed 2017-09-05 by mertenats