SamZorSec/Open-Home-Automation #17: Issues with uploading sketch for ha_mqtt_sensor_dht22.ino

AssuredDust640, 2017-09-07:
This is my first go at using MQTT in HA and trying to upload the sketch to the ESP8266 so please be understanding of my ignorance.

Made the minor adjustments to your ino file for my wifi network and MQTT details. Plus the topic name and the sketch verifies perfectly.

When I hook up the NodeMCU board to the pc to upload I get a rather long error file (below). Any advice would be greatly appreciated.

Arduino: 1.8.4 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (144K SPIFFS), ck, Disabled, None"

Archiving built core (caching) in: C:\Users\ADMINI~1.ASU\AppData\Local\Temp\arduino_cache_134493\core\core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_dio,UploadSpeed_115200,FlashSize_1M144,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____7246470ddf479558ce5125d1daf92f35.a
Sketch uses 240089 bytes (27%) of program storage space. Maximum is 876528 bytes.
Global variables use 33296 bytes (40%) of dynamic memory, leaving 48624 bytes for local variables. Maximum is 81920 bytes.
C:\Users\Administrator.ASU-HP7900-DS\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9/esptool.exe -vv -cd ck -cb 115200 -cp COM3 -ca 0x00000 -cf C:\Users\ADMINI~1.ASU\AppData\Local\Temp\arduino_build_821035/sketch_sep07a.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2013 bytes of data
	read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2013 bytes of data
	read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2013 bytes of data
	read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2013 bytes of data
	read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2013 bytes of data
	read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2013 bytes of data
	read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2013 bytes of data
	read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2013 bytes of data
	read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2013 bytes of data
	read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mertenats, 2017-09-07:
Hi @AssuredDust640,
I assume that the compilation went fine, so the problem might come from the upload process.

- Do you have something connected to D8? If yes, momentary unplug it.
- Do you have chosen the right board? For the NodeMCU board, you have to select the "NodeMCU v1.0" option (Tools>Board).

Sam

AssuredDust640, 2017-09-07:
Sam, thanks for your reply.

Yes, the compilation went fine, it’s just at the upload stage I get the errors. I’ve played with changing the board between the NodeMCU and generic ESP8266 and both caused issues, but as it is a NodeMCU, I’ll stick with that going forward.

When you mention that is something connected to the DB, should I have the DHT sensor NOT wired up when I upload? Should the DB just be by itself for this process and then re-wire the sensor into place?

Thanks again for your assistance.

Jonathan

mertenats, 2017-09-08:
Hi,
I experienced issues when something was connected to D8 during the upload process (logic level probably set to HIGH).. You should maybe try to unplug/your NodeMCU to the computer, press the flash/reset button before uploading, etc. Do you have another NodeMCU? Have you ever suceeded in uploading something to the board?

Sam

AssuredDust640, 2017-09-08:
Sadly this is the only NodeMCU I have and I’ve yet to upload anything to it but I’m just about to try unhooking everything, resetting it and trying again. Thanks for your assistance, it’s been very helpful.

Regards,

Jonathan

AssuredDust640, 2017-09-08:
All good, thanks again.

Sketch uploaded and I can see it on my router. Just need to wait and see if the info comes through to HA now but that should be the least of my worries.

Received 2 x Sonoff wifi switches today so on to those next.

Can’t express my thanks enough.

Regards,

Jonathan

closed 2017-09-08 by mertenats