SamZorSec/Open-Home-Automation #18: Sorry, me again with another question

AssuredDust640, 2017-09-09:
After getting your assistance with the upload to my NodeMCU, I am very appreciative.

However, it seems I'm not getting any output to my MQTT broker from the setup. The only difference between mine and yours is I'm using a DHT11 sensor not a DHT22.

Other then the one place you directly mention it (and which I amended it to 11 from 22) is there anything else I might need to change to make this work or do you think I might have issues with the DB or sensor themselves.

Jonathan

AssuredDust640, 2017-09-09:
Ok, just shoot me for my stupidity. I'm using a 3 pin DHT11, data is the first pin not the middle one.

Most likely have destroyed the sensor by plugging data into the positive wires but at least I know better now.

Sorry to have bothered you again for nothing

mertenats, 2017-09-15:
@AssuredDust640,
No worries.

Sam

closed 2017-09-15 by mertenats