SamZorSec/Open-Home-Automation #20: PIR help

samabsalom, 2017-10-02:
Hi Sam,
as always i still love what you've done. it basically all runs my house. i was wondering whether you could point me in the right direction with altering the PIR file?

1. House alarms do the opposite of the pir sensors you have made the code for. house alarms are normally closed going open on movement so it actually breaks the 5v instead of sending 5v. how would i accommodate this in the .ino file?
2. How can i scale up the file? simply duplicating everything and allocating a number to it i.e. pirstate1 pirstate2 etc doesnt work. im assuming because of the loop. could you advise me how to go about doing this? 

any little bit of help would be greatly appreciated
Thanks,
Sam

mertenats, 2017-10-02:
Hi @samabsalom,
Thanks for your message.
1. Use interrupts and trigger an interruption when the signal is FALLING (movement detected) and RISING (movement ended). Example [here](https://github.com/mertenats/Open-Home-Automation/blob/master/ha_mqtt_binary_sensor_door/ha_mqtt_binary_sensor_door.ino). Or use the example with the PIR sensor and reverse the logic. 
2. I'm developing a device with multiple sensors (still in development): https://github.com/mertenats/Open-Home-Automation/tree/dev/ha_mqtt_multisensor

Sam

SammyPlant, 2018-02-13:
Hi Sam,
I too like how robust your code is but you still haven't answered his second question as your multi sensor code is for different sensors but what about multiple PIR sensors as in my case i would like to use 6 pir's with your code and thats not straight forward as the previous sam suggested. So the question still stands how do I report the status of multiple PIR sensors to the mqtt broker.

Sam

mertenats, 2018-02-16:
Hi,
You may find an answer in this [sketch](https://github.com/mertenats/Open-Home-Automation/blob/master/ha_mqtt_binary_sensor_nfc_scanner/ha_mqtt_binary_sensor_nfc_scanner.ino) , which contains an array of structures, one structure per device (or sensor in your case).

Define a structure for each PIR sensor  :
```
typedef struct {
  ...
  bool    isDiscovered;
  bool    toNotify;
  char    mqttTopic[48];
} NFCTag;
```

Loop through the array of structures and publish the new state of the modified PIR sensor(s) :
 
```
void loop() {
  connectToMQTT();
  mqttClient.loop();

  ...

  yield();

  for (uint8_t i = 0; i < NB_OF_NFC_TRACKED_TAGS; i++) {
    if (NFCTags[i].toNotify) {
      if (NFCTags[i].isDiscovered) {
        publishToMQTT(NFCTags[i].mqttTopic, MQTT_PAYLOAD_ON);
      } else {
        publishToMQTT(NFCTags[i].mqttTopic, MQTT_PAYLOAD_OFF);
      }
      NFCTags[i].toNotify = false;
    }
  }
  
  yield();
}
```

And define two interrupt routines for each sensor, triggered with a FALLING/RISING edge and change the `isDiscovered` state accordingly.

Sam

closed 2018-02-19 by mertenats