SamZorSec/Open-Home-Automation #21: Change %brightness  to number LUX

danielaradu2016, 2017-10-13:
Hy
I want to ask you if i can change %brightness to number brightness. (ex: 234 lux) for photocell.
Thx a lot

mertenats, 2017-10-24:
I'll update the example soon. Until then, you can look at the conversion [here](https://github.com/mertenats/Open-Home-Automation/blob/master/ha_mqtt_multisensor/MultiSensor.cpp).

Read the LDR:
```
#if defined(LDR_SENSOR)
  static unsigned long lastLdrSensorMeasure = 0;
  if (lastLdrSensorMeasure + LDR_MEASURE_INTERVAL <= millis()) {
    lastLdrSensorMeasure = millis();
    uint16_t currentLdrValue = analogRead(LDR_SENSOR);
    if (currentLdrValue <= this->_ldrValue - LDR_OFFSET_VALUE || currentLdrValue >= this->_ldrValue + LDR_OFFSET_VALUE) {
      this->_ldrValue = currentLdrValue;
      evt = LDR_SENSOR_EVT;
      return;
    }
  }
#endif
```

Convert the measured voltage in lux:
```
#if defined(LDR_SENSOR)
uint16_t MultiSensor::getLux(void) {
  // http://forum.arduino.cc/index.php?topic=37555.0
  float voltage = this->_ldrValue * LDR_VOLTAGE_PER_ADC_PRECISION;   
  return 500 / (LDR_RESISTOR_VALUE * ((LDR_REFERENCE_VOLTAGE - voltage) / voltage));
}
#endif
```

Constants:
```
#define LDR_OFFSET_VALUE                25
#define LDR_MEASURE_INTERVAL            15000 // [ms]
#define LDR_REFERENCE_VOLTAGE           3.3   // [v]
#define LDR_ADC_PRECISION               1024  // 10 bits
#define LDR_VOLTAGE_PER_ADC_PRECISION   LDR_REFERENCE_VOLTAGE / LDR_ADC_PRECISION
#define LDR_RESISTOR_VALUE              10.0  // [kOhms]
```

danielaradu2016, 2017-11-24:
I just made the multisensor and i have the same problem with the photocell. When i have a lot of luz he can show me only 4 lux. a little luz ... 0 lux.
I have 10koms and 3,3v
How can i calibrate¿¿
THX A LOT

danielaradu2016, 2017-11-24:
Can be my sensor. i use this ... https://es.aliexpress.com/item/Free-Shipping-10pcs-TEMT6000-Light-Sensor-Professional-TEMT6000-Light-Sensor-Module/32584176331.html?spm=a219c.10010108.1000016.1.593202c5AKvSE&isOrigTitle=true

thx

danielaradu2016, 2017-11-24:
Is must be GL5516¿¿¿¿ like this.......https://es.aliexpress.com/store/product/wholesale-new-original-50PCS-GL5516-5516-Light-Dependent-Resistor-LDR-5MM-Photoresisto-photosensitive-sensor-For-Arduino/1773295_32410482019.html?spm=a219c.12010108.1000016.1.7d8b4d61cRcDes&isOrigTitle=true

thx

mertenats, 2018-05-11:
Updated code to convert the AnalogRead value to lux:

```
uint16_t MultiSensor::getLux(void) {
  // http://forum.arduino.cc/index.php?topic=37555.0
  // https://forum.arduino.cc/index.php?topic=185158.0
  float volts = this->_ldrValue * REFERENCE_VOLTAGE / ADC_PRECISION;
  float amps = volts / LDR_RESISTOR_VALUE;
  float lux = amps * 1000000 * 2.0;
  return uint16_t(lux);
}
```

cf. the multi sensor sketch.

closed 2018-05-11 by mertenats