SamZorSec/Open-Home-Automation #27: Controlling 4 lights

PavanSN, 2018-01-22:
how can i control many lights using mqtt light.ino file, ive tried many ways of modifying the  code but it didnt work... please help me to fix..

gabrielcsilva, 2018-01-23:
I have reached a functional code that could control as many lamps as the esp can hold (theoretically). I had to change mostly the functions to receive a position counter of an array of lamps.

The code still have some points that need improvement, like in the lines 38 and 39 with an automatic generation of the topics according to the number of lamps; and the line 45, because the code doesn't work with an array of lamps using PROGMEM for a reason that I can't tell.

If there's something wrong about what I did, i apologize @mertenats 

[Test-_4_lamps_MQTT.zip](https://github.com/mertenats/Open-Home-Automation/files/1654175/Test-_4_lamps_MQTT.zip)

mertenats, 2018-02-15:
Hi @PavanSN,
Someone asked me the same question a few months ago. Just look at this [issue]/https://github.com/mertenats/Open-Home-Automation/issues/1), a example is provided.

Sam

closed 2018-02-15 by mertenats