SamZorSec/Open-Home-Automation #32: Add SHT30 Temp/Humidity Sensor support

pidiet, 2018-04-01:
Hey,

this one is also great, small and more precise than dht22. And it has a wemos shield as well: https://wiki.wemos.cc/products:d1_mini_shields:sht30_shield

**Notes:**
Wemos' provided library is hinky and not recommended.
This firmware uses ClosedCube's library instead: https://github.com/closedcube/ClosedCube_SHT31D_Arduino (not closed at all!)
Source: https://github.com/boneskull/shtshow

pidiet, 2018-04-08:
This project here uses the sht30 sensor:
- https://github.com/boneskull/shtshow
based on this library here:
- https://github.com/closedcube/ClosedCube_SHT31D_Arduino

Maybe it's a good starting point?

mertenats, 2018-04-22:
Hi @pidiet,
One sensor has just been ordered.

rradar, 2018-05-22:
@mertenats thank's so much!

Just builded my first multi sensor with a sht30! Just works! Great!