SamZorSec/Open-Home-Automation #35: binary sensor door/button VS multi sensor door/button

pidiet, 2018-04-08:
Hello,

i just switched over from using the binary sensor door (ha_mqtt_binary_sensor_door) over to use the multisensor (ha_mqtt_multisensor) for the same propose plus more (pir sensor)

I found some difference:

magnet/reed/door sensor with the same wiring (d1 and ground) results:
- binary sensor: works like a charm
- multi sensor: also works but **open/closed is vice versa?**

button with same wiring (wemos button shield on d3 and ground) without any resistor:
- binary sensor: button press forces reboot of the esp board
- multi sensor: works like a charm

And the pir/motion sensor on the multi sensor sketch also works like a charm.

Beside that I like the "online" state information from the multi sensor! Very nice feature!

Would be awesome to "tune" the multisensor sketch that it's possible to use more/different types of sensor. For example:
- light sensor TEMT6000 #31 
- temperature DS18B20 #34  
- temperature and humidity SHT30 #32 
- temperature, humidity and pressure BME280 #33   
- and maybe even an actor like a RGB LED #30 

Any idea how I can correct my open/closed problem with the my door sensor? Thank's!

rradar, 2018-04-09:
Like the idea! Having the multi sensor with modules you can choose and wire your hardware! :+1:

pidiet, 2018-04-13:
**How can I change the (wrong) open/closed state of my magnet/reed/door sensor?**

mertenats, 2018-05-11:
@pidiet Just invert the payload sent to the MQTT broker:

Replace the following lines in config.h:
```
#define MQTT_PAYLOAD_ON   "ON"
#define MQTT_PAYLOAD_OFF  "OFF"
```

by 

```
#define MQTT_PAYLOAD_ON   "OFF"
#define MQTT_PAYLOAD_OFF  "ON"
```

closed 2018-05-11 by mertenats