SamZorSec/Open-Home-Automation #37: Can't compile the newest version from multisensor

rradar, 2018-05-18:
I'm using Arduino 1.8.2

and get the following output (did already a second git clone and a manual grab just to be sure no files where corrupted over the air)

Error message is the following one (files untouched just renamed the example to config.h)

```
In file included from sketch/MultiSensor.h:5:0,
                 from sketch/MultiSensor.cpp:2:
sketch/MultiSensor.cpp: In member function 'void MultiSensor::init()':
sketch/config.h:5:41: error: 'D3' was not declared in this scope
 #define DOOR_SENSOR                     D3
                                         ^
sketch/MultiSensor.cpp:55:11: note: in expansion of macro 'DOOR_SENSOR'
   pinMode(DOOR_SENSOR, INPUT_PULLUP);
           ^
sketch/config.h:16:44: error: 'D7' was not declared in this scope
 #define MOTION_SENSOR_PIN                  D7
                                            ^
sketch/MultiSensor.cpp:60:11: note: in expansion of macro 'MOTION_SENSOR_PIN'
   pinMode(MOTION_SENSOR_PIN, INPUT_PULLUP);
           ^
sketch/config.h:56:39: error: 'D2' was not declared in this scope
 #define SHT_SCL_PIN                   D2
                                       ^
sketch/MultiSensor.cpp:75:14: note: in expansion of macro 'SHT_SCL_PIN'
   Wire.begin(SHT_SCL_PIN, SHT_SDA_PIN);
              ^
sketch/config.h:60:41: error: 'D1' was not declared in this scope
 #define BUTTON_SENSOR                   D1
                                         ^
sketch/MultiSensor.cpp:82:11: note: in expansion of macro 'BUTTON_SENSOR'
   pinMode(BUTTON_SENSOR, INPUT);
           ^
sketch/MultiSensor.cpp: In member function 'void MultiSensor::handleEvt()':
sketch/config.h:16:44: error: 'D7' was not declared in this scope
 #define MOTION_SENSOR_PIN                  D7
                                            ^
sketch/MultiSensor.cpp:102:23: note: in expansion of macro 'MOTION_SENSOR_PIN'
       if (digitalRead(MOTION_SENSOR_PIN) != this->_motionState) {
                       ^
sketch/MultiSensor.cpp: In member function 'bool MultiSensor::_readDoorState()':
sketch/config.h:5:41: error: 'D3' was not declared in this scope
 #define DOOR_SENSOR                     D3
                                         ^
sketch/MultiSensor.cpp:223:22: note: in expansion of macro 'DOOR_SENSOR'
   return digitalRead(DOOR_SENSOR);
                      ^
exit status 1
```

An older version worked with my setup out of the box?

per1234, 2018-05-18:
What do you have selected in the **Tools > Board** menu?

mertenats, 2018-05-21:
Which sensors are you using at the same time?

rradar, 2018-05-22:
Problem was related to the host and/or the board. Changed host computer and board and it worked directly!

closed 2018-05-22 by rradar